Deftaudio/Midi-boards #6: MIDI Merge 2in->10out loses "note off" messages in merge mode

ericlathrop, 2021-11-05:
When I play my keyboard through the device with ports set to "merge" mode, MIDI "note off" messages go missing, and the note plays forever. This happens even when only a single device is sending notes to input port 1.

When I set the jumpers to "separate" mode everything seems to work correctly, and notes end when they should.

ericlathrop, 2021-11-05:
I was unable to find the source code for the merge IC. Could you add it to the repo, and maybe I can help debug it?

Deftaudio, 2021-11-05:
Hi the source code for Merger IC is not going to be posted. They shipped pre-programmed and run the only firmware version. Does it happen on the second MIDI input as well?