Deftaudio/Midi-boards #7: Add OSC-midi over ethernet for teensy 4.1 breakout possible?

Keeze, 2021-11-30:
Hello,

I am using a Midibox Seq4 and like to use the Teensy 4.1 MIDI Breakout board 8in / 8out USB host as a breakout for a syntrack. I have an ethernet connexion in my Midibox and it talks OSC midi. Is it possible to add the ethernet breakout to the teensy and use it to connect to my midibox?

The sequencer has 4 OSC ports defined. So it would be nice to route each port to 2 of the serial midiports...

Love to hear your thoughts..

Cheers,
Kees

Deftaudio, 2021-12-01:
Hi Kees, 
You can use other Teensy pins, including for ethernet or USB expansion, all pins are available for dupont patching around teensy. just make sure that they don't interfere with Serial 1-8 ports.

Keeze, 2021-12-04:
Hmm, I have a ethernetboard soldered and now looking for examples how to implement itand no luck yet. I am a very basic coder (aka C&P-coder) and need some example code. Anyone here who can point me in the right direction?

Deftaudio, 2021-12-04:
Hi, 
Sorry, I can't be helpful here. 
The Teensy forum would be the best place to ask this question. 
https://forum.pjrc.com/

Keeze, 2021-12-05:
Ok, thanks

closed 2021-12-05 by Keeze