Deftaudio/Midi-boards #12: Midi Teensy 4.1 8x8 only shows 4x4

circuitcreature, 2025-03-10:
On a new build, using MacOS, only 4 midi in and 4 midi out ports are showing.

Arduino IDE Compile settings:

Tools > USB Type: MIDIx16

Deftaudio, 2025-03-11:
Hi, I think you should check again. 
I've just compiled it on the latest Sequoia 15.4 beta 2 and I don't see any issues. # # 

[image]

You may try to disconnect Teensy, delete a MIDI device in audio/midi setup and connect it again. You may also just take a hex file and upload it directly to your Teensy https://github.com/Deftaudio/Midi-boards/blob/master/MIDITeensy4.1/Firmware/interface_8x8/interface_8x8.ino.TEENSY41.hex

If problem continues to persist, report it and PJRC forum.

circuitcreature, 2025-03-11:
Thank you, removing the 4x4 device from the audio/midi setup and reconnecting solved the issue.

closed 2025-03-11 by circuitcreature