Deftaudio/Midi-boards #13: Coding [Eurorack MIDI Teensy 5x5]

rthorntn, 2025-05-13:
Hi,

Sorry to bother you.

I'm really interested in buying some Eurorack MIDI Teensy 5x5 modules as the hardware looks perfect but I'm worried about having to program them, I am not a coder.  Is there a way the code is arranged so you can set variables for each port, for setting up merge, split and filter, [a long shot but] has there been a configuration UI created?

Can the module be a USB host for MIDI, if so, is there code for this?

Can it have multiple merge groups simultaneously, say:

 - IN ports 1,2,3 merge to OUT port 4
 - IN ports 4,5 merge to OUT port 5

if so, is there code for this?

Can it filter or remap notes, channels or messages, if so, is there code for this?

Can it merge on some ports and split on others, if so, is there code for this?

Thanks.

rithma, 2025-05-13:
It’s really easy to understand code. It’s all broken up into sections with
full comments.  I am not a coder by any means (I can print hello world and
make an LED blink but that’s about it).   I was able to change some of the
ports to pass through midi data etc with relative ease.

Rithma Mobile

Deftaudio, 2025-05-13:
If you glance the example code https://github.com/Deftaudio/Midi-boards/blob/master/Eurorack_MIDI_Teensy_5x5/interface_5x5_audio/interface_5x5_audio.ino
you'll commented sections, such as lines 73-78
/* MIDI1.send(type, data1, data2, channel);
      MIDI2.send(type, data1, data2, channel);
      MIDI3.send(type, data1, data2, channel);      
      MIDI4.send(type, data1, data2, channel);  
      MIDI5.send(type, data1, data2, channel);  
   */ 
This is exaclty what you can un-comment for your needs to start splitting or merging signals. There is a bit of learning needed for sure. This project is meant for that.

flyingzebra, 2025-09-29:
I just ordered the Eurorack MIDI Teensy 5x5 Interface Merger breakout.
Since I plan to hookup a TFT touch screen to this module using the DMA feature, I will likely have to sacrifice/disconnect pin 12 from Teensy 4.0 to the board, which currently comes wired to the audio output.   It's a pity but I guess it won't be too hard to reassign this audio output pin to another pin on the MCU. I want the display to receive something more useful than a sinusoid :-D

Deftaudio, 2025-09-29:
Hi, you don't even need to disconnect anything. Just don't plug audio jack. There are two capacitors C8/C9 in series. You can keep or omit those. 
There is no way to reassing those pins unfortunately. They are coming of the MSQ audio ports.